ros/meta-ros #1480: ERROR setuptools_scm._file_finders.git listing git files failed - pretending there aren't any

uestccaixukun, 2025-07-16:
I encountered this error while compiling qt-gui-cpp. I looked up some information and it seems to be a common problem when executing "colcon build". But how can I solve this in yocto?
`ERROR setuptools_scm._file_finders.git listing git files failed - pretending there aren't any`

mark-bchoi, 2025-07-17:
Hi @uestccaixukun ,

Can you describe yocto version and ros2 version you tried ?

mark-bchoi, 2025-07-22:
I missed a patch file. It was created by @robwoolley  which I found a PR but, I could not remember exact PR number. 
Can you please try it?

`sip-build-cross-platform-support.patch`

```
sip-build: Cross platform build support

Projects may use the PyQtBuilder in pyproject.toml to automatically
generate the build files and execute the build.

PyQtBuilder/builder.py invokes buildable.get_module_extension() to
create rules in QMake project files for building Python extensions
for the target.

However, this determines the module extension using the importlib module.
This provides the extension suffix used for Python modules that
may be imported. However, this is incorrect when running Python
in a cross-build environment.  For OpenEmbedded, the correct
extension suffix for the target may be obtained by referring to
the sysconfig EXT_SUFFIX configuration variable.

Upstream-Status: Inappropriate [oe build specific]

--- a/sipbuild/buildable.py
+++ b/sipbuild/buildable.py
@@ -150,14 +150,8 @@ class BuildableModule(BuildableFromSourc
         if self.project.py_platform == 'win32':
             return '.pyd'

-        from importlib.machinery import EXTENSION_SUFFIXES
-
-        if self.uses_limited_api:
-            for s in EXTENSION_SUFFIXES:
-                if '.abi3' in s:
-                    return s
-
-        return EXTENSION_SUFFIXES[0]
+        import sysconfig
+        return sysconfig.get_config_var('EXT_SUFFIX')

 class BuildableBindings(BuildableModule):
```

uestccaixukun, 2025-07-22:
> I missed a patch file. It was created by [@robwoolley](https://github.com/robwoolley) which I found it from PR but, I could not remember exact PR number. Can you please try it?
> 
> `sip-build-cross-platform-support.patch`
> 
> ```
> sip-build: Cross platform build support
> 
> Projects may use the PyQtBuilder in pyproject.toml to automatically
> generate the build files and execute the build.
> 
> PyQtBuilder/builder.py invokes buildable.get_module_extension() to
> create rules in QMake project files for building Python extensions
> for the target.
> 
> However, this determines the module extension using the importlib module.
> This provides the extension suffix used for Python modules that
> may be imported. However, this is incorrect when running Python
> in a cross-build environment.  For OpenEmbedded, the correct
> extension suffix for the target may be obtained by referring to
> the sysconfig EXT_SUFFIX configuration variable.
> 
> Upstream-Status: Inappropriate [oe build specific]
> 
> Signed-off-by: Rob Woolley <rob.woolley@windriver.com>
> 
> --- a/sipbuild/buildable.py
> +++ b/sipbuild/buildable.py
> @@ -150,14 +150,8 @@ class BuildableModule(BuildableFromSourc
>          if self.project.py_platform == 'win32':
>              return '.pyd'
> 
> -        from importlib.machinery import EXTENSION_SUFFIXES
> -
> -        if self.uses_limited_api:
> -            for s in EXTENSION_SUFFIXES:
> -                if '.abi3' in s:
> -                    return s
> -
> -        return EXTENSION_SUFFIXES[0]
> +        import sysconfig
> +        return sysconfig.get_config_var('EXT_SUFFIX')
> 
> 
>  class BuildableBindings(BuildableModule):
> ```
> 
> This patch is for sip.

thanks,
I've seen this patch in the issue (https://github.com/ros/meta-ros/issues/1424). At that time I applied robwoolley's modifications regarding sip and pyqt5, but failed to resolve the compilation issue. Based on your previous modifications, I reapplied the above sip and pyqt5 changes, but the result seems identical to not applying them - while compilation passes, it still appears to not provide the binding (I'm unsure whether the absence of libqt_gui_cpp_sthiboken.so and libqt_gui_cpp_sip.so would affect Python bindings, such as providing the python_qt_binding.QtSvg module). Your previous solution seems to have circumvented the sip compilation issue rather than solving it.

robwoolley, 2025-09-04:
The fixes for pyqt5 should all be merged upstream.  Can you confirm which layers and commits you are using?  Something like the following output from bitbake would be ideal:

```
Build Configuration:                                                                                   BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"                                                                  NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi5"
DISTRO               = "ros2"                                                                          DISTRO_VERSION       = "2.humble.2025-08-15.3-0"                                                       TUNE_FEATURES        = "aarch64 crypto cortexa76"                                                      TARGET_FPU           = ""                                                                              DISTRO_NAME          = "Robot Operating System 2 (ROS 2) with OpenEmbedded Linux"
ROS_DISTRO           = "humble"
ROS_VERSION          = "2"                                                                             ROS_PYTHON_VERSION   = "3"                                                                             meta-clang           = "scarthgap:6691e4bc47e20b903dc3dd7684685086d2468b84"                            meta-clang-revival   = "scarthgap:f34e150f6def4f7a9226bad517b6f6b4dcadb516"                            meta-filesystems                                                                                       meta-gnome                                                                                             meta-initramfs                                                                                         meta-multimedia                                                                                        meta-networking                                                                                        meta-oe                                                                                                meta-perl                                                                                              meta-python                                                                                            meta-webserver                                                                                         meta-xfce            = "scarthgap:e8fd97d86af86cdcc5a6eb3f301cbaf6a2084943"                            meta                 = "scarthgap:7a59dc5ee6edd9596e87c2fbcd1f2594c06b3d1b"                            meta-python-ai       = "scarthgap:57dba41efc2a938f10a4120f6723dcb7baa68f26"                            meta-qt5             = "scarthgap:554fc20c326a9b205169331644f0cb471619e63c"                            meta-raspberrypi     = "scarthgap:aaf976a665daa7e520545908adef8a0e9410b57f"                            meta-intel-realsense = "scarthgap:67d11b1aacb456ccf7cdfa1127b8fa6594cf7ab4"                            meta-ros-common                                                                                        meta-ros2                                                                                              meta-ros2-humble     = "scarthgap:bbf7d8db818ab40b8e9ed10cf702bbae5ac7f2cd"                            meta-zenoh           = "scarthgap:0ec94ef54136a90e5bc6c79d283b3121a7d21cb5"                            
```

robwoolley, 2026-05-07:
Closing as fixed.  Please reopen if it continues to be a problem.