ros/meta-ros #1496: rcl-yaml-param-parser configure failure

seb-832, 2025-07-28:
Hi

On the latest scarthgap revision, the recipe `rcl-yaml-param-parser` fails to run the configure task with the following output:

```
| CMake Error at /app/tmp/work/cortexa76-actia-linux/rcl-yaml-param-parser/5.3.9-1/recipe-sysroot/usr/lib/cmake/yaml/yamlConfig.cmake:16 (message):
|   File or directory
|   /app/tmp/work/cortexa76-actia-linux/rcl-yaml-param-parser/5.3.9-1/recipe-sysroot/lib/cmake/yaml/yamlTargets.cmake
|   referenced by variable yaml_TARGETS does not exist !
```

This is possibly related to commit: c2a975fd77f70558e4e67f2480d7fd3fe0cf1897 because we have `usrmerge` set in the `DISTRO_FEATURES` that affects the `libdir` value.

I guess there is another recipe that still use the hardcoded `lib` path instead of `${libdir}` somewhere.
Any idea where to solve that ?

Thanks.

Zeug555, 2025-07-29:
Same issue here, I am intested if you have a solution !

Zeug555, 2025-08-01:
@vermaete I don't know for @seb-832 but for me it doesn't work with the scarthgap version and the humble ROS2 distribution.

robwoolley, 2025-08-01:
Thanks, I'll take a look into it.

mvorobiev, 2025-08-07:
Hi

Do you have any updates on this issue? I can't build with the heading commit on the Scarthgap branch. If I rollback to the previous commit, it builds just fine.

robwoolley, 2025-08-10:
I finished merging the July Sync for Rolling into Scarthgap: https://github.com/ros/meta-ros/commits/scarthgap-next/

The desktop-full variant builds successfully.  I plan to merge it into the scarthgap branch once I have given it some "soak time".

There is a fix to libyaml that should resolve the problem with libyaml-vendor and rcl-yaml-param-parser: https://github.com/ros/meta-ros/commit/81f9a7955319d6d44e811653bfe7c5afc7e1166a

pberginkonsult, 2025-08-11:
Why are `‎meta-ros-common/recipes-support/libyaml/libyaml_0.2.5.bb` present in scarthgap branch when the same version is present in oe-core `meta/recipes-support/libyaml/libyaml_0.2.5.bb`?

robwoolley, 2025-08-11:
Hi @pberginkonsult the commit message describes the motivation behind it: https://github.com/ros/meta-ros/commit/cb333721bda56165eebb9b836c6698535b22eeca

The short version is that we need to use CMake instead of autotools.

robwoolley, 2025-09-04:
I can confirm that rcl-yaml-param-parser builds successfully with the latest changes on scarthgap+humble.

Please feel free to file another issue if you continue to experience problems with it.

closed 2025-09-04 by robwoolley