ros/meta-ros #1533: [scarthgap][humble] cartographer build failed with absl

Nettoysfkf, 2025-08-17:
My environment:
board: rockchip rk3588
yocto and ros2 branch:  scarthgap

My test:
Testing Navigation2 with turtlebot3 in gazebo,  now navigation2 and gazebo are ok for my build.
But when build the cartographer:
```
In file included from /workdir/ddk/2025-project/ddk-github-next/build-rk3588-ros/tmp/work/cortexa76-cortexa55-poky-linux/cartographer/2.0.9004-1/git/cartographer/common/task.h:24,
                 from /workdir/ddk/2025-project/ddk-github-next/build-rk3588-ros/tmp/work/cortexa76-cortexa55-poky-linux/cartographer/2.0.9004-1/git/cartographer/common/task.cc:17:
/workdir/ddk/2025-project/ddk-github-next/build-rk3588-ros/tmp/work/cortexa76-cortexa55-poky-linux/cartographer/2.0.9004-1/git/cartographer/common/thread_pool.h:67:58: error: expected ';' at end of member declaration
   67 |   std::weak_ptr<Task> Schedule(std::unique_ptr<Task> task)
      |                                                          ^
      |                                                           ;
/workdir/ddk/2025-project/ddk-github-next/build-rk3588-ros/tmp/work/cortexa76-cortexa55-poky-linux/cartographer/2.0.9004-1/git/cartographer/common/thread_pool.h:68:22: error: 'mutex_' has not been declared
   68 |       LOCKS_EXCLUDED(mutex_) override;
      |                      ^~~~~~
/workdir/ddk/2025-project/ddk-github-next/build-rk3588-ros/tmp/work/cortexa76-cortexa55-poky-linux/cartographer/2.0.9004-1/git/cartographer/common/thread_pool.h:68:7: error: ISO C++ forbids declaration of 'LOCKS_EXCLUDED' with no type [-fpermissive]
   68 |       LOCKS_EXCLUDED(mutex_) override;
      |       ^~~~~~~~~~~~~~

```
According to AI report,  i tried to disable thread safety in cartographer, but still failed with abover error.

It seems that the cartographer does't compatible with absl thread safety that the  [abseil version in meta-oe was 20240116](https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-devtools/abseil-cpp/abseil-cpp_20240116.3.bb?h=scarthgap)

vermaete, 2025-08-17:
@Nettoysfkf does this patch on Rolling help you?  https://github.com/ros/meta-ros/blob/master/meta-ros2-rolling/recipes-bbappends/cartographer/cartographer/use-newer-abseil-api.patch

Nettoysfkf, 2025-08-19:
I fnd a resolution for  [link error with abseil](https://github.com/abseil/abseil-cpp/discussions/1586). 
It suggests to set abseil c++ to 17, but it means targets need to set c++ 17 that depends abseil, such as protobuf and so on.

Nettoysfkf, 2025-08-19:
That patch works, so  closed this.

closed 2025-08-19 by Nettoysfkf